gauravmm/HierarchicalPcb #2: fails to replicate first layout

huegene, 2023-10-17:
Hi, 
when I use this plugin, the first instance of the schematic is notplaced/replicated in the layout and remains untouched (not placed and not routed) the subsequent layouts are recognised and placed and routed

huegene, 2023-10-17:
I traced the error to the DlgHPCBRun.__init__() 
for what ever reason the first addition of anitem fails.
reassigning the parent item during the first run 
seems to fix the problem 
            item: wx.TreeListItem = self.treeApplyTo.AppendItem(
                parent=parent_item, text=row_text, data=sheet
            )
            if (self.treeApplyTo.GetCount()==0):
                parent_item: wx.TreeListItem = (
                    sheet.parent.list_ref or self.treeApplyTo.GetRootItem()
                )
                item: wx.TreeListItem = self.treeApplyTo.AppendItem(
                    parent=parent_item, text=row_text, data=sheet
                )

gauravmm, 2023-10-17:
Huh, that's really odd. Could you please share a minimum PCB that exhibits this problem?

Thanks!

gauravmm, 2023-11-19:
Hey, I messed with this a little bit and I couldn't replicate it. If you can send me a minimal PCB that exhibits this problem, please let me know and I'll reopen the issue.

hrueger, 2024-03-13:
Hi @gauravmm,
I've just managed to run into this. Here you can find a reproduction project: 
[first-layout-does-not-replicate.zip](https://github.com/gauravmm/HierarchicalPcb/files/14593158/first-layout-does-not-replicate.zip)

Just open the `testhira.kicad_pcb` file and see for yourself. Thanks in advance!

gauravmm, 2024-03-14:
Hey, thanks! I'll take a peek at this over the weekend.

hrueger, 2024-03-14:
@gauravmm thanks! We've tried @huegene's fix and it works fine. We just don't understand why ;-)

OfficialDyray, 2024-03-19:
I've run into this problem aswell. One thing I figured out is if you edit the .heirpcb.json and manual change the first uuid's checked to true, it starts replicating again. This makes me think there is a GUI issue with showing the first entry (hira).
```
  "sheet": {
    "": {
      "checked": false
    },
    "/37288e56-ec97-4563-8d14-ca8a67be2c8f": {
      "checked": true <-------------------------------- //CHANGED THIS TO TRUE
    },
    "/b5fc7067-d6a0-4ae8-afd1-af302add3973": {
      "checked": true
    },
    "/c8933244-f192-49b4-aa2a-eddd57cad00a": {
      "checked": true
    }
  }
```

hrueger, 2024-03-22:
Interestingly, while [making the treeview sort items correctly](https://github.com/gauravmm/HierarchicalPcb/commit/b1042d392f24cc3ea3e2c03361c0ea6780fae0cb) it works just fine without @huegene's fix....

gauravmm, 2024-03-30:
Thanks for looking into this! I'll adopt @hrueger's fix and then close this.

hrueger, 2024-03-30:
You might rather look into @OfficialDyray as his proposal is probably the real fix and not just a workaround like @huegene's changes.

gauravmm, 2024-03-30:
I looked at it a little and I concur. Closing this and merging that.

closed 2024-03-30 by gauravmm