gauravmm/HierarchicalPcb #3: Not compatible with KiCAD 7.99 (nightly)

StickySli, 2023-10-26:
The project is available in the 7.99 plugin repository but the plugin does not show up in the PCBNew editor. There might be something going on.

gauravmm, 2023-10-27:
I'll check over the weekend, thanks for the heads-up.

gauravmm, 2023-11-19:
Okay, so I took a look at it and figured out the source of the bug. The proximal error is that pcbnew no long has `FP_TEXT`. I'll figure out how pcbnew now handles text input and fix it shortly.

For documentation, I debugged the loading process by adding code to `__init__.py` to log the plugin loading.  The output is:

```
  File "...\placement.py", line 112, in PositionTransform
    src: Optional[pcbnew.FP_TEXT],
                  ^^^^^^^^^^^^^^
AttributeError: module 'pcbnew' has no attribute 'FP_TEXT'
```

The logging is:

```python
import traceback
from pathlib import Path
with Path("C:\\Users\\User\\Desktop\\hierarchical.log").open("w") as f:
    f.write("Hello World\n")
    try:
        from .hplugin import HierarchicalPCBPlugin
        HierarchicalPCBPlugin().register()
    except Exception as e:
        f.write(str(e))
        f.write(traceback.format_exc())
    finally:
        f.write("Goodbye World\n")
```

nikolasbrendes, 2024-01-24:
I Got the same Error and just removed the Function body in placement.py Line 110:
`Path: /home/MYUSERNAME/.local/share/kicad/8.0/3rdparty/plugins/com_github_gauravmm_hierarchicalpcb` 

```
def footprint_text(
        self,
        src, #: Optional[pcbnew.FP_TEXT],
        dst, #: Optional[pcbnew.FP_TEXT],
        dst_fp: Optional[pcbnew.FOOTPRINT] = None,
    ):
        return
        """Move dst to the same position as src, but relative to the anchor_mutate."""

        if type(src) == type(dst) == pcbnew.FP_TEXT:
            src.SetLayer(dst.GetLayer())
            dst.SetTextThickness(src.GetTextThickness())
            dst.SetTextWidth(src.GetTextWidth())
            dst.SetTextHeight(src.GetTextHeight())
            dst.SetItalic(src.IsItalic())
            dst.SetBol
     .
     .
     .
     .
```
After the changes the Plugin got loaded, but crashed when opened.

Another Error occurred in hdata.py Line 272 + 273:
```
                sheet_file = Path(fp.GetProperty("Sheetfile"))
                sheet_name = fp.GetProperty("Sheetname")
```
It seems like they removed the GetProperty() function.

I found another function in the FOOTPRINT Class that seems promising:  `GetSheetfile()` and `GetSheetname()`
```
                sheet_file = Path(fp.GetSheetfile())
                sheet_name = fp.GetSheetname()
``` 
I replaced the Lines 272 and 273 with the code above and i got the Plugin back running!

Hope this helps :)

gauravmm, 2024-02-17:
That helps a lot, thank you! I'll update the code and merge a bunch of changes soon -- likely next weekend.

chatelao, 2024-02-28:
It doesn't run with KiCAD 8.0 launched last week. Would love to use it, sounds great.

btw: love your keyboard design too.

gauravmm, 2024-03-02:
Hey, thanks! I'll tackle KiCAD 8.0 compatibility tomorrow, for now this person has a proposed fix: https://github.com/gauravmm/HierarchicalPcb/pull/6

OmegaRogue, 2024-03-15:
Hows this looking?

OfficialDyray, 2024-03-30:
Looking good so far:

- Supports cloning zones and drawings ( This includes edge cuts/images/text/silkscreen symbols)
- properly mirroring text on the front and back of the pcb. 
- bug fixes.

OmegaRogue, 2024-04-03:
anything that isnt working in v8 yet?

gauravmm, 2024-04-28:
This version works pretty well: https://github.com/gauravmm/HierarchicalPcb/tree/feature-kicad8

The current state is being tracked here: #7 . The bottleneck to releasing this is some text rotation ( solved in #12, not merged yet ), and some via issues ( #18, just merged). Before I release it as compatible, it also needs to be tested against this large sample project: https://github.com/gauravmm/AngloDox/

Its going to take me another week or two, I'm unfortunately busy with real life.

gauravmm, 2024-05-05:
KiCAD 8 support is now live on master! Thanks to @OfficialDyray @hruger @anemograph @ferris and others who helped, we couldn't have figured this out without you.

I'll be submitting a patch to the official repository soon.

closed 2024-05-05 by gauravmm