gauravmm/HierarchicalPcb #9: 🐛 replication fails on some PCBs (KiCad 8.0.0)

lukas-runge, 2024-03-14:
After making the plugin work with KiCad 8 by using the `feature-kicad8` branch and implementing [this workaround](https://github.com/gauravmm/HierarchicalPcb/issues/2#issuecomment-1766131050) to fix replication on the first layout I discovered a new bug. I have one component that just does not replicate.
Here is a minimal example to reproduce the issue: [non-working-component-demo.zip](https://github.com/gauravmm/HierarchicalPcb/files/14605070/non-working-component-demo.zip)
Maybe you can have a look into it @gauravmm.

Huge thanks in advance! 🙏

Kind regards
Lukas

lukas-runge, 2024-03-14:
@gauravmm this issue was an oopsie on my side. I had not updated my components pcb from the schematic with relinking via designators enabled. Thus I had a uuid mismatch between schematic and pcb parts. 🙈

Closing the issue as it never was related to the plugin in the first place. 😇

closed 2024-03-14 by lukas-runge

gauravmm, 2024-03-15:
No problem, thanks for following up!