gauravmm/HierarchicalPcb #10: `AppendNet` does not exist in Kicad 8

hrueger, 2024-03-21:
Hi @gauravmm,
we discovered another small bug: in KiCad 8, [this `AppendNet`](https://github.com/gauravmm/HierarchicalPcb/blob/d5125aa21821ecf65acaf0655bfd54a782ba6bd6/placement.py#L292) is no longer available.
You can see it here:
[Kicad 7 API](https://docs.kicad.org/doxygen-python-7.0/classpcbnew_1_1NETINFO__LIST.html)
[Kicad 8 API](https://docs.kicad.org/doxygen-python-8.0/classpcbnew_1_1NETINFO__LIST.html)

When commenting out this line, it still works though. Maybe it is not needed anymore?

Best
Hannes

hrueger, 2024-03-22:
will be fixed by #7 (sorry for the noise, hadn't read the comments over there)

gauravmm, 2024-03-30:
No problem, thanks for the report!

closed 2024-03-30 by gauravmm