gauravmm/HierarchicalPcb #17: Doesn't work with KiCad 8

JLA4444, 2024-04-14:
I as going to use this great looking plugin but it doesn't load correctly on KiCad 8

gauravmm, 2024-04-28:
Thanks for the interest! This version works pretty well: https://github.com/gauravmm/HierarchicalPcb/tree/feature-kicad8

The current state of KiCAD 8 compatibility is being tracked here: #7. Mainly, we need some minor bugs fixed and a little more testing, especially with larger projects.

closed 2024-04-28 by gauravmm

JLA4444, 2024-04-28:
I have a large project to test on so tell me when you need it tested.

By large i mean many different subschematic not many different instances of the same subschematic

gauravmm, 2024-04-28:
Much appreciated! I'll reach out to you soon, likely by next weekend.

gauravmm, 2024-05-05:
@JLA4444 Its ready for you to test it, if you can clone it to your plugins directory. Let me know if you need help with that.

(It will take a few days to hit the actual KiCAD repo.)