gauravmm/HierarchicalPcb #19: Implicit nets disapear when editing subcircuit pcb

cgerlach42-2, 2024-05-06:
Good timing. Was searching for something like this plugin just 6h after it got it's KiCad 8.0 release. Thank You!

I see an issue when updating the pcb from the schematic for the subcircuit. Implicitly named nets do not have a name and thus do not get airwires in the pcb.

This is the sub-schematic when opened in the base project:
[image]
The selected net between R3 and R7 gets the name Net-(R3-Pad2).

When I open the schematic in the project for the the sub-schematic (to do the layout) it looks like this:
[image]
See the /\<NO NET\> in the status bar. The airwire will also not appear when updating the pcb.

Workaround to explicitly name all nets seems to work.
-----------
Application: KiCad Schematic Editor x86_64 on x86_64

Version: 8.0.2-8.0.2-0~ubuntu22.04.1, release build

Libraries:
	wxWidgets 3.2.1
	FreeType 2.11.1
	HarfBuzz 2.7.4
	FontConfig 2.13.1
	libcurl/7.81.0 OpenSSL/3.0.2 zlib/1.2.11 brotli/1.0.9 zstd/1.4.8 libidn2/2.3.2 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.43.0 librtmp/2.3 OpenLDAP/2.5.17

Platform: Ubuntu 22.04.4 LTS, 64 bit, Little endian, wxGTK, X11, ubuntu-xorg, x11

Build Info:
	Date: Apr 28 2024 21:38:46
	wxWidgets: 3.2.1 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.6.3
	Curl: 7.81.0
	ngspice: 42
	Compiler: GCC 11.4.0 with C++ ABI 1016

Build settings:

gauravmm, 2024-05-13:
Thanks for the good example and the detailed bug report! We struggled a bit to figure out how to deal with nets in the KiCAD 8 release, and (unfortunately) you've encountered one of the remaining rough edges.

I think I know how this happened -- I'll take a look at it this weekend.

cgerlach42-2, 2024-05-17:
My guess here is that kicad can not name the net in the sub schematic as the automatic names have the part designator in their names. These are of cause only available in the instantiation of the sheet.

OfficialDyray, 2024-05-17:
One thing I noticed was when you  re-annotate the sub-pcb (In the sub-pcb file) , the NO_NET disappears. Or if you change the reference number of any of the symbols, it claims the net.