gauravmm/HierarchicalPcb #20: Zones get wrong net assigned

cgerlach42-2, 2024-05-07:
The README states, zones are not supported. I guess this is somehow outdated (looking at the functionality and the commits).
Still, they seem to get wrong net names assigned. I am not familiar with KiCad internals, but looking at the .kicad_pcb file there seem to be two assignments: net and net_name.
In the resulting file 'net_name' seems to have the right name but missing the hierarchical part (its '/netname' instead of '/subcircuitX/netname').  And 'net' refers to a completely wrong net number (different net than the one referred in 'net_name'.
This results in different net names being displayed int the status bar and the properties editor when the zone is highlighted. And of cause wrong connections.

gauravmm, 2024-05-13:
Thanks for the bug report, and for noticing that the hierarchical part is missing. I'll take a crack at it this weekend. 

If you have the chance, I'd appreciate it if you could upload a small dummy project with a zone and explain exactly what the plugin gets wrong? I'm a relative newbie at PCBs and have never used the Zone feature in KiCAD.

Thanks!

OfficialDyray, 2024-05-16:
Pull request added. I did some basic testing, which passed the sniff test. I might revisit nets in the future to pin down a more durable way of finding net names.

gauravmm, 2024-05-16:
Thanks! I'll take a look at it.

cgerlach42-2, 2024-05-17:
Just checked the zone handling: seems to do the trick. Checked with local net names, global net names and power nets. At least in my small test project.
I guess you don't need the test project anymore (just created it today)...

gauravmm, 2024-05-17:
I'll definitely take the test project if you have it! I'll add it to our set of test/example cases so we can use it to check future software updates.

cgerlach42-2, 2024-05-17:
[kicad_hierachical_test.zip](https://github.com/gauravmm/HierarchicalPcb/files/15346536/kicad_hierachical_test.zip)
Quick and dirty test project.
Issue remaining is the unnamed local net (as mentioned in my other issue)