gauravmm/HierarchicalPcb #30: Layers without additionnal components cannot be used

xDuinoRails, 2024-12-17:
IS
- PCBs of schematics containing no additionall components (but labels etc.) cannot be used
- They get a strange alpha-numeric names (see below e.g. "b23a7a80")

SHOULD
- Sheets / PCBs without components should  be usable too.

WORKAROUND
- Dummy components like a testpoint solve the problem.

Hierarchical view:
[image]
Page view with schematics:
[image]

<3 <3 <3 the plug-in

OfficialDyray, 2024-12-22:
Components provide an easy way to track position and rotation. Other board features like drawings don't provide absolute position and rotation, which makes things tricky. You can create a special "origin" footprint specifically for translation without any board features.
[Origin.zip](https://github.com/user-attachments/files/18220427/Origin.zip)

closed 2024-12-27 by gauravmm

chatelao, 2024-12-27:
Thanks a lot for the feedback and the amazing product.