gauravmm/HierarchicalPcb #31: Cannot enforce layout of intermedate nesting

olk-jbozzo, 2025-02-06:
Hello,

This issue may just be a misunderstanding of the Readme of the repository:

> Instead of laying out everything in your final file, you can define a sub-PCB next to your sub-schematic file, and HierarchicalPCB will automatically force the sub-PCB to be laid out exactly like that in the final PCB. There's no depth limitation to the nesting -- in fact, we encourage organizing your PCB layouts for maximum reusability.

What I gather from it is that description is that I could enforce the PCB layout of any level of the nested subsheets. But in practice I have only been able to enforce the layout of leaf nodes.

Example
====

## Setup

1. I create a root.kicad_pro project with 3 levels of nesting, root > layer1 > layer2
  
 - On layer2 I placed a single SMD resistor
[image]

- On layer1 I placed two layer2
[image]

- On root I placed two layer1
[image]

2. I close the previous project and create a layer2.kicad_pro project in the same folder as root.kicad_pro. I edit and only save the pcb diagram with the single resistor placed.
[image]

3. I close the previous project and create layer1.kicad_pro project in the same folder as root.kicad_pro. I edit and only save the pcb diagram with the two instances of the resistors connected in an angle (I've used HierarchicalPCB to enforce the layout of layer2, even if it is only the resistor)
[image]

3. I close the previous project and open root.kicad_pro, then go to the PCB, use Update PCB from Schematic (F8) and click on the HierarchicalPCB icon.

## What the problem looks like

The HierarchicalPCB dialog only lets me enforce the layout of layer2 PCB.
[image]

## What I expect

The HierarchicalPCB let me choose between enforcing the layout of layer2 or layer1 PCB (image edited to show expectation)
[image]

I've attached the projects folder so it may be easier to debug the issue (if that is the case).
[root.zip](https://github.com/user-attachments/files/18693983/root.zip)

olk-jbozzo, 2025-02-07:
I've looked into the code and it seems that this is a limitation from the plugin. The plugin requires that every intermediate pcb to have at least one footprint that is not from it's children. This is commented in hdata.py:

```python

def get_sheet_hierarchy(
    board: pcbnew.BOARD, basedir: Path
) -> Tuple[SchSheet, Dict[Path, PCBRoom]]:
    """Infer the sheet hierarchy from footprint data.

    While this should be better handled by examining the schematics, we can't yet do that in KiCad.
    Note that this cannot find sheets that are not referenced by at least one footprint.
    """
```

If I add a footprint to layer1, for example a testpoint connected to nothing, then the plugin works exactly as expected.

closed 2025-02-07 by olk-jbozzo

gauravmm, 2025-06-23:
Hey, thanks for figuring this out and documenting it!